acalatrava/openhaystack-firmware #1: Build firmware fails

mowtschan, 2021-05-12:
I tried to build the firmaware, but got the error, so can I build the firmware on macos?
```
NRF_MODEL=nrf51 make
```

```
BUILD OPTIONS:
  SoftDevice  s130
  SDK         11
  nRF         nrf51822
  RAM         16 kB
  FLASH       256 kB
  Board       BOARD_CUSTOM

  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/toolchain/system_nrf51.c
  CC        ble_stack.c
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//ccTJhp9G.s: Assembler messages:
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//ccTJhp9G.s:27: Error: SVC is not permitted on this architecture
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//ccTJhp9G.s:46: Error: SVC is not permitted on this architecture
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//ccTJhp9G.s:65: Error: SVC is not permitted on this architecture
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//ccTJhp9G.s:84: Error: SVC is not permitted on this architecture
make: *** [_build/ble_stack.o] Error 1
```

acalatrava, 2021-05-12:
Yes of course. In fact I’m building it on MacOS. It looks like a compiler error, do you have `arm-none-eabi-gcc` installed?

mowtschan, 2021-05-12:
I installed it like that:

```
brew tap PX4/homebrew-px4
brew install gcc-arm-none-eabi
```

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
```

acalatrava, 2021-05-12:
Please issue a `make -n` and post it

mowtschan, 2021-05-12:
`NRF_MODEL=nrf51 make -n`

```
BUILD OPTIONS:
  SoftDevice  s130
  SDK         11
  nRF         nrf51822
  RAM         16 kB
  FLASH       256 kB
  Board       BOARD_CUSTOM
 
mkdir -p _build/
echo "  CC       " ble_stack.c
arm-none-eabi-gcc -mthumb -mcpu=cortex-m0 -march=armv6-m -L ../../nrf5x-base/make/../make/ld -T ../../nrf5x-base/make/../make/ld/gcc_nrf51_s130_2.0.0_16_256.ld -Wl,--gc-sections -Wl,-Map=_build/openhaystack-alternative_s130.Map -Os -fomit-frame-pointer -DBLE_STACK_SUPPORT_REQD -DSOFTDEVICE_PRESENT -DCENTRAL_LINK_COUNT=0 -DPERIPHERAL_LINK_COUNT=1 -DBLEADDR_FLASH_LOCATION=0x0003FFF8 -std=gnu99 -c -mthumb -mcpu=cortex-m0 -march=armv6-m -Wall -Wextra -Wno-unused-parameter -Werror=return-type -DNRF51 -DBOARD_CUSTOM -DNRF51822 -DSDK_VERSION_11 -DSOFTDEVICE_s130 -s -ffunction-sections -fdata-sections -DS130 -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/toolchain/gcc/ -I. -I../../nrf5x-base/make/../advertisement/ -I../../nrf5x-base/make/../devices/ -I../../nrf5x-base/make/../devices/tcmp441/ -I../../nrf5x-base/make/../devices/tcmp441/libqrencode/ -I../../nrf5x-base/make/../lib/ -I../../nrf5x-base/make/../lib/simple_logger/ -I../../nrf5x-base/make/../lib/simple_logger/chanfs -I../../nrf5x-base/make/../lib/simple_logger/mem-ffs -I../../nrf5x-base/make/../peripherals/ -I../../nrf5x-base/make/../services/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/ant_fs/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/bootloader_dfu/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/button/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/crc16/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/crc32/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/ecc/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/experimental_section_vars/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/fds/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/fifo/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/fstorage/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/gpiote/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/hardfault/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/hci/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/ic_info/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/led_softblink/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/low_power_pwm/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/mailbox/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/mem_manager/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/pwm/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/scheduler/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/sensorsim/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/sha256/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/simple_timer/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/slip/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/timer/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/trace/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/twi/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/uart/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/adc/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/ble_flash/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/clock/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/common/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/delay/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/gpiote/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/hal/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/lpcomp/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/ppi/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/pstorage/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/pstorage/config/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/radio_config/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/rng/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/rtc/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/sdio/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/spi_master/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/spi_slave/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/swi/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/timer/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/twi_master/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/uart/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/wdt/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/validation/ -I../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components [... cut ...]
echo "  BIN      " _build/openhaystack-alternative_s130.bin
arm-none-eabi-objcopy -Obinary _build/openhaystack-alternative_s130.elf _build/openhaystack-alternative_s130.bin
echo "  LST      " _build/openhaystack-alternative_s130.lst
arm-none-eabi-objdump --disassemble-all --source --disassembler-options=force-thumb -C --section-headers _build/openhaystack-alternative_s130.elf > _build/openhaystack-alternative_s130.lst
echo "  SIZE     " _build/openhaystack-alternative_s130.elf
arm-none-eabi-size _build/openhaystack-alternative_s130.elf
```

acalatrava, 2021-05-12:
It looks ok.... what Mac are you using and what MacOS version?

acalatrava, 2021-05-12:
But are you on a M1 or Intel?

mowtschan, 2021-05-12:
`sysctl -a | grep machdep.cpu.brand_string`
```
machdep.cpu.brand_string: Intel(R) Core(TM) i7-8850H CPU @ 2.60GHz
```

acalatrava, 2021-05-13:
I'm unable to reproduce the issue, and I'm on almost the same version as you (11.2.3). If I remember correctly I downloaded the gcc-arm-none-eabi from here https://developer.arm.com/-/media/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-mac.pkg?revision=96309e24-064b-4aaf-81fc-9a4f2f15417e&la=en&hash=FB3A659BD38CEFCE2CF4EA78CBD95DA44ACE8AD9
Please try that.

mowtschan, 2021-05-13:
no luck :-(

```
➜  openhaystack-alternative git:(main) which arm-none-eabi-gcc
/Applications/ARM/bin/arm-none-eabi-gcc
➜  openhaystack-alternative git:(main) arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

➜  openhaystack-alternative git:(main) rm -rf _build
➜  openhaystack-alternative git:(main) NRF_MODEL=nrf51 make
BUILD OPTIONS:
  SoftDevice  s130
  SDK         11
  nRF         nrf51822
  RAM         16 kB
  FLASH       256 kB
  Board       BOARD_CUSTOM

  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/toolchain/system_nrf51.c
  CC        ble_stack.c
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//cccWgUhd.s: Assembler messages:
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//cccWgUhd.s:27: Error: SVC is not permitted on this architecture
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//cccWgUhd.s:46: Error: SVC is not permitted on this architecture
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//cccWgUhd.s:65: Error: SVC is not permitted on this architecture
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//cccWgUhd.s:84: Error: SVC is not permitted on this architecture
/var/folders/zs/rw8wlnrx6kq39jhksv6bt2lw0000gn/T//cccWgUhd.s:103: Error: SVC is not permitted on this architecture
make: *** [_build/ble_stack.o] Error 1
```

acalatrava, 2021-05-13:
Hmmm, it seems I'm using a way older version
```
 arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```

Please try the same version from here
https://developer.arm.com/-/media/Files/downloads/gnu-rm/5_4-2016q3/gcc-arm-none-eabi-5_4-2016q3-20160926-mac.tar.bz2?revision=06ff216c-f946-4b4a-bea1-22a5e09502ca?product=GNU%20Arm%20Embedded%20Toolchain%20Downloads,64-bit,,Mac%20OS%20X,5-2016-q3-update

mowtschan, 2021-05-13:
WOW, works! Thanks! 🙏 😊 

```
➜  openhaystack-alternative git:(main) arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

➜  openhaystack-alternative git:(main) rm -rf _build
➜  openhaystack-alternative git:(main) NRF_MODEL=nrf51 make
BUILD OPTIONS:
  SoftDevice  s130
  SDK         11
  nRF         nrf51822
  RAM         16 kB
  FLASH       256 kB
  Board       BOARD_CUSTOM

  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/toolchain/system_nrf51.c
  CC        ble_stack.c
  CC        main.c
  CC        openhaystack.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/softdevice/common/softdevice_handler/softdevice_handler.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_util_platform.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/common/nrf_drv_common.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/delay/nrf_delay.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_error.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_error_weak.c
  AS        ../../nrf5x-base/make/../startup/startup_nrf51.s
  LD        _build/openhaystack-alternative_s130.elf
  HEX       _build/openhaystack-alternative_s130.hex
  BIN       _build/openhaystack-alternative_s130.bin
  LST       _build/openhaystack-alternative_s130.lst
  SIZE      _build/openhaystack-alternative_s130.elf
   text	   data	    bss	    dec	    hex	filename
   5092	   2180	    160	   7432	   1d08	_build/openhaystack-alternative_s130.elf
```

closed 2021-05-13 by mowtschan

acalatrava, 2021-05-13:
Using newer `arm-none-eabi-gcc` is fixed now https://github.com/acalatrava/openhaystack-firmware/commit/05495d111827bb47218090f005764fe1e0fbdec2

commit 05495d11 (2021-05-13): This will fix #1 and allow using newer arm-none-eabi-gcc