acalatrava/openhaystack-firmware #2: Flash firmware with openocd works but there is no advertisement so far

mowtschan, 2021-05-13:
I flashed nrf51 device with the firmware `apps/openhaystack-alternative/compiled/nrf51_firmware.bin` (without any modifications), I used openocd and stlink adapter. Flashing process worked without any issues but there is no advertisement so far, am I missing something?
```
Open On-Chip Debugger
> halt
target halted due to debug-request, current mode: Handler SVCall
xPSR: 0x6100000b pc: 0x000091f4 msp: 0x20003f48
> nrf51 mass_erase
nRF51822-QFAA(build code: H0) 256kB Flash, 16kB RAM
Mass erase completed.
A reset or power cycle is required if the flash was protected before.
> program /openhaystack-firmware/apps/openhaystack-alternative/compiled/nrf51_firmware.bin
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Adding extra erase range, 0x0001cc70 .. 0x0001cfff
** Programming Finished **
> resume
> exit
```

acalatrava, 2021-05-13:
Keep in mind that the advertisement is set to 5000 msec so you may miss some of the advertisements. For iPhone I saw that the eBeacon app (https://apps.apple.com/es/app/ebeacon-ble-scanner/id730279939) works better than others. Sometimes you may need to wait 20-30 seconds to see the first advertisement though.

mowtschan, 2021-05-13:
I'm using the BlueSee app for macOS which is even better than the iPhone app

acalatrava, 2021-05-13:
I installed BlueSee app and I don't see the device but I can see it on eBeacon on iPhone.... In fact I only see 2 devices on BlueSee while eBeacon shows me 6, including my TV... Not sure why

mowtschan, 2021-05-13:
Did you install BlueSee app on mac or on iPhone?

mowtschan, 2021-05-13:
anyway with BlueSee I can exactly see which device is mine (I can see the public key of the device) but with all other tools I cannot see it

acalatrava, 2021-05-13:
On Mac. Ok, I forgot to uncheck the filter to show only connectable devices... After some time the device appears:
[image]

mowtschan, 2021-05-13:
do you have original openhaystack firmware or is it your alternative firmware? is it on nf51 device?

[image]

acalatrava, 2021-05-13:
This is my alternative firmware just downloaded from releases and flash it on a nRF51 device. The only difference with you is that I flash it using JLink since I don't have STLink. But this is the same file you flash and on the same device....

acalatrava, 2021-05-13:
It took some time to appear though, like 30-40 seconds

mowtschan, 2021-05-13:
I can successfully build, flash (with stlink) and see advertising (with bluesee app) from example `nrf5x-base/apps/s130-beacon`:

[image]

but still no luck with your firmware :-/

acalatrava, 2021-05-13:
Did you try to change the advertising interval?

mowtschan, 2021-05-13:
`apps/openhaystack-alternative/main.c`

[image]

mowtschan, 2021-05-13:
it works! :-) (for my device nRF51822-QFAA) 
if I change `NRF_CLOCK_LFCLKSRC`:

[image]

closed 2021-05-13 by mowtschan

acalatrava, 2021-05-13:
Yeah! I thought about that but I couldn’t reply earlier. That’s because your board has no external crystal so you need to use the internal one. You can change this on the board_custom file instead or create a custom board file for your board.

acalatrava, 2021-05-13:
Forced to not use an external crystal by default https://github.com/acalatrava/openhaystack-firmware/commit/1524dea6d9c57065447b7b9dec21872a9f07e234