acalatrava/openhaystack-firmware #3: No Makefile

jaekim24, 2021-05-13:
I'm trying to flash the nRF52 chip but I'm having a hard time. I'm new to all this, especially github, I tried running the "make build" command and I got this error message. 
[image]

acalatrava, 2021-05-13:
You need to fetch the submodule. Go to the root directory and issue  
```
git submodule init
git submodule update
```

jaekim24, 2021-05-13:
that fixed it !!

jaekim24, 2021-05-13:
now when I enter "make build" I get this error at the bottom about "make: arm-none-eabi-gcc: No such file or directory" [image]

acalatrava, 2021-05-13:
You need to install this
https://developer.arm.com/-/media/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-mac.pkg?revision=96309e24-064b-4aaf-81fc-9a4f2f15417e&la=en&hash=FB3A659BD38CEFCE2CF4EA78CBD95DA44ACE8AD9

jaekim24, 2021-05-13:
I downloaded it but I still get the same error @acalatrava

acalatrava, 2021-05-13:
I think you need to add it to your path. Try this

TOOLCHAIN_PATH=/Applications/ARM/bin make build

jaekim24, 2021-05-13:
I still get the same error @acalatrava 
[image]

acalatrava, 2021-05-13:
Sorry I missed the latest /

```
TOOLCHAIN_PATH=/Applications/ARM/bin/ make build
```

jaekim24, 2021-05-14:
I have this error now, I checked and I don't have a makefile. Do I have to make one?
[image]

jaekim24, 2021-05-14:
new error after I typed ```TOOLCHAIN_PATH=/Applications/ARM/bin/ make build
``` : 
<img width="1006" alt="Screen Shot 2021-05-14 at 1 18 48 AM" src="https://user-images.githubusercontent.com/62858192/118225057-56b5b900-b452-11eb-8b86-13787b70eed8.png">

mowtschan, 2021-05-14:
`brew install binutils`

jaekim24, 2021-05-14:
I still get the same error even after installing binutils

mowtschan, 2021-05-14:
@jaekim24 can post the output of the command `which objcopy` ?

jaekim24, 2021-05-14:
"objcopy not found" @mowtschan

acalatrava, 2021-05-14:
What about `which gobjcopy`

jaekim24, 2021-05-14:
"gobjcopy not found"

mowtschan, 2021-05-14:
@jaekim24 looks like it is missing in your `PATH`, can you run `export PATH="/usr/local/opt/binutils/bin:$PATH"` and try again?

jaekim24, 2021-05-14:
I think that worked, so with the new binary that's been created I use that to flash on to the compiled directory ?
[image]

jaekim24, 2021-05-16:
how do I flash the binary to the nRF52 chip?

acalatrava, 2021-05-16:
If you have a JLink you can just issue a `make flash`

jaekim24, 2021-05-16:
what about the public advertisement key, Do I have to get that from the open haystack app?

jaekim24, 2021-05-17:
im having this error where the "Target connection not established yet but required for command." 
[image]

jaekim24, 2021-05-17:
[image]

jaekim24, 2021-05-17:
when I tried a new chip it did this:
michaelhunt@nicolaskerr openhaystack-alternative % TOOLCHAIN_PATH=/Applications/ARM/bin/ make flash
BUILD OPTIONS:
  SoftDevice  s132
  SDK         11
  nRF         nrf52832
  RAM         64 kB
  FLASH       512 kB
  Board       BOARD_SIMPLE
 
mkdir -p _build/
printf "r\n" > _build/flash.jlink
printf "loadfile _build/openhaystack-alternative_s132.hex \nr\ng\nexit\n" >> _build/flash.jlink
mkdir -p _build/
printf "r\nsavebin _build/downloaded_test.bin 0x1000 0x10\nexit\n" > _build/test-softdevice.jlink
mkdir -p _build/
SEGGER J-Link Commander V6.88a (Compiled Nov 18 2020 15:09:53)
DLL version V6.88a, compiled Nov 18 2020 15:09:35

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04
Hardware version: V7.00
S/N: 20090928
License(s): RDI,FlashDL,FlashBP,JFlash,GDBFull
VTref=3.300V
Target connection not established yet but required for command.
Device "NRF52" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
Found SW-DP with ID 0x2BA01477
Unknown DP version. Assuming DPv0
Scanning AP map to find all available APs
AP[2]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
AP[1]: JTAG-AP (IDR: 0x02880000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 003BB001 ITM
ROMTbl[0][4]: E0040000, CID: B105900D, PID: 000BB9A1 TPIU
ROMTbl[0][5]: E0041000, CID: B105900D, PID: 000BB925 ETM
Cortex-M4 identified.
Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.

Opening binary file for writing... [_build/downloaded_test.bin]
Reading 16 bytes from addr 0x00001000 into file...O.K.

Script processing completed.

16+0 records in
16+0 records out
16 bytes transferred in 0.000131 secs (122016 bytes/sec)
Files _build/downloaded_test.bin and _build/softdevice_test.bin differ
[... 175 lines cut ...]
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
Cannot connect to target.

Target connection not established yet but required for command.
Device "NRF52" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
Cannot connect to target.

Script processing completed.

michaelhunt@nicolaskerr openhaystack-alternative %

jaekim24, 2021-05-17:
when I try flashing the new chip again this shows up : 

michaelhunt@nicolaskerr openhaystack-alternative % TOOLCHAIN_PATH=/Applications/ARM/bin/ make flash
BUILD OPTIONS:
  SoftDevice  s132
  SDK         11
  nRF         nrf52832
  RAM         64 kB
  FLASH       512 kB
  Board       BOARD_SIMPLE
 
mkdir -p _build/
printf "r\n" > _build/flash.jlink
printf "loadfile _build/openhaystack-alternative_s132.hex \nr\ng\nexit\n" >> _build/flash.jlink
mkdir -p _build/
printf "r\nsavebin _build/downloaded_test.bin 0x1000 0x10\nexit\n" > _build/test-softdevice.jlink
mkdir -p _build/
SEGGER J-Link Commander V6.88a (Compiled Nov 18 2020 15:09:53)
DLL version V6.88a, compiled Nov 18 2020 15:09:35

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Cannot connect to J-Link.

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Failed to open DLL

Script processing completed.

16+0 records in
16+0 records out
16 bytes transferred in 0.000139 secs (115110 bytes/sec)
Files _build/downloaded_test.bin and _build/softdevice_test.bin differ
BUILD OPTIONS:
  SoftDevice  s132
  SDK         11
  nRF         nrf52832
  RAM         64 kB
  FLASH       512 kB
  Board       BOARD_SIMPLE
 
# Write to NVMC to enable erase, do erase all, wait for completion. reset
mkdir -p _build/
printf "w4 4001e504 2\nw4 4001e50c 1\nsleep 100\nr\nexit\n" > _build/erase-all.jlink
mkdir -p _build/
JLinkExe -device nrf52 -if swd -speed 1000  _build/erase-all.jlink
SEGGER J-Link Commander V6.88a (Compiled Nov 18 2020 15:09:53)
DLL version V6.88a, compiled Nov 18 2020 15:09:35

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Cannot connect to J-Link.

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Failed to open DLL

Sleep(100)

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Failed to open DLL

Script processing completed.

# Write to NVMC to enable write. Write mainpart, write UICR. Assumes device is erased.
mkdir -p _build/
printf "w4 4001e504 1\nloadbin \"_build/s132_nrf52_2.0.0_softdevice.bin\" 0\nr\ng\nexit\n" > _build/flash-softdevice.jlink
[... 303 lines cut ...]
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
Cannot connect to target.

Target connection not established yet but required for command.
Device "NRF52" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
Cannot connect to target.

Script processing completed.

michaelhunt@nicolaskerr openhaystack-alternative %

acalatrava, 2021-05-18:
It seems that you find the same issue as me while loading the firmware. Please try `make e104install` instead. You may need to disconnect and connect VCC while this command is being executed (and maybe several times until it triggers the flash)

jaekim24, 2021-06-02:
thank you I will try this @acalatrava

jaekim24, 2021-06-02:
how do I add my advertisement key ?

jaekim24, 2021-06-04:
[image]
im getting this error @acalatrava

jaekim24, 2021-06-04:
[image]
is it talking about this hex file

closed 2022-12-30 by jaekim24