acalatrava/openhaystack-firmware #6: Trying to add a new board into the project

VladutLP, 2021-05-28:
Hey there @acalatrava, 

I'm trying to add the board **PCA10059** from the latest sdk. 
This is for a [nrf52840 dongle](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-Dongle).

I've added `pca10059.h` in the root dir and linked it into `boards.h`, compiled everything with a base64 key from UI but it does not want to show up 😞 

I had to modify `pca10059.h` and add `NRF_CLOCK_LFCLKSRC` part from another board.

I've checked the BLE advertisements and they are showing up every 5 sec just like in the code. Is there a way I can double check the actual key from the advertisement using an Android phone?

- Decoding the address with `xxd -P` turns back: 

```
base64 -D <<< uVXys0qVPiqsqVzZhb+IoLGZSWA9ToGCa1tDaA== | xxd -p 
b955f2b34a953e2aaca95cd985bf88a0b19949603d4e81826b5b4368
```

- The advertisement from BLE Scanner Android app: 

[image]

- Dongle Address: `CF:46:46:4C:49:4E`

(I have 4 other original AirTags near the devboard and they show up constantly on Find My. The reciever is my Macbook 2019 on bigSur 11.5 and a couple of neighbouring iPhones)

VladutLP, 2021-05-28:
Fixed. I'll write a detailed how to with the PR. Basically the advertisement key was not replaced in the hex. I think this is only for the bin.

closed 2021-05-28 by VladutLP

bruvv, 2022-07-19:
@VladutLP I am willing to buy the nRF52840 chip too, can you let me know how you fixed it? I do not see a PR from you.

foxmzslashX, 2023-05-27:
Sorry for resurrecting this thread, but im afraid i have a similar problem.

It is strange, because I already flashed a couple of nRF51822s with different public keys on each of them, and they are working fine. Now, a few weeks later i cannot build a working firmware (the build succeeds without errors), the keys are not changing in the advertisement. 
Actually it doesn't matter what public key i use for building the firmware, the advertised result always starts with `CF:46:46:4C:49:4E` too.

@VladutLP Have you managed to solve this?