acalatrava/openhaystack-firmware #9: nRF51822 with/without oscillator build process and energy consumption

sebi5361, 2021-06-12:
I installed your openhaystack firmware alternative on an [AliExpress beacon](https://www.aliexpress.com/item/32826502025.html) and it works great! Thank you.

I also ordered a [YJ-15013-NRF51822 module](https://www.aliexpress.com/item/32965118537.html) and it is equipped with an oscillator.
Q: Is there a build parameter to specify the use of an oscillator when building the nRF51822 firmware (or should I create a new board definition)?

In the product description it is written: _"Oscillator, in Bluetooth low-power applications, does not require an external 32kHz crystal, can be sectioned"_
Q: Do you think having no oscillator consumes less energy indeed?
Q: Do you think I have to remove physically the oscillator from the board or using the firmware with no oscillator should prevent the microcontroller from powering the oscillator?

sebi5361, 2021-06-14:
I found my answers regarding power consumption [here](https://devzone.nordicsemi.com/f/nordic-q-a/14214/how-much-current-consumption-when-we-use-internal-32k-rc).
In a nutshell: an external crystal is more energy efficient than the internal RC oscillator.
I'll adjust a board definition file according to my needs.
Am closing this issue.

closed 2021-06-14 by sebi5361

lanrat, 2022-06-25:
@sebi5361 I have the same beacon. Can you share your board definition file?

Thanks.