acalatrava/openhaystack-firmware #12: Any tips on how to physically connect to suggested beacons?

EdanZwick, 2022-06-11:
(This is not an issue with the code, just asking since you might know the answer)
Hey,

Thanks for this work, seems very cool!

I'm just getting into BLE as a hobby and don't have much experience with hardware, so this is probably a silly question, but what tools do I need to program the hardware you suggested such as:

"AliExpress beacon" which can be purchased here https://www.aliexpress.com/item/32826502025.html

I have to solder the power supply and data pins for programming, right?
I have access to a soldering station, but I think that this might be too small to solder, no?

I have a similar beacon by Raytac:
https://www.amazon.com/dp/B081GYNM2G?ref_=cm_sw_r_cp_ud_dp_RFBYEBARPW5J6KK9CCJV
But its too small for me to solder and seems like it requires some more professional gear.

Thanks for the help!
Edan

acalatrava, 2022-06-12:
Hi Edan!

Yes, soldering may be tricky. Specially on that module since the pads are very close together. I used a very similar module and I don’t even solder to program it! I used some hook clips I had around like these: https://a.aliexpress.com/_vpQ1qK

with a little care you will manage to hook into the right pads without soldering. 

However after that I had to connect a CR2032 battery so I had to solder the VCC pad and for the GND I used the metal on the cover. You should use a very tiny lead soldering wire and a small tip and if I could do it I’m sure you will too ;)

EdanZwick, 2022-06-13:
Thanks! Will share what I end up with :)

defic, 2022-06-22:
Thanks for the tips 😊  Just made an order for "AliExpress beacons". I've only flashed microcontrollers with usb interface before. Can you recommend me a low cost programming adapter from Aliexpress that I can use to flash the beacon?

mowtschan, 2022-06-22:
> Thanks for the tips 😊 Just made an order for "AliExpress beacons". I've only flashed microcontrollers with usb interface before. Can you recommend me a low cost programming adapter from Aliexpress that I can use to flash the beacon?

I used that one(is not available anymore but just search for "st link v2"):
https://www.amazon.de/gp/product/B086TWZNMM

how to connect: https://gist.github.com/mowtschan/d1956b4634a84698ab3a96472f710610

vekexasia, 2022-08-16:
hey @mowtschan thanks for the gist. Very helpful. I don't understand `base64 -D <<< your_base64_key_from_openhaystack_app | xxd -p | cut -c13-18` what is this used for. Can you please clarify?

mowtschan, 2022-08-17:
@vekexasia , with the bluesee app (or similar) you see a lot of devices but you don't know which one is yours. 
I used that command to extract some part of the key to be able to track/recognise my device with the bluesee app.

closed 2023-01-19 by acalatrava

mundeczek, 2026-03-16:
> Hi Edan!
>
> Yes, soldering may be tricky. Specially on that module since the pads are very close together. I used a very similar module and I don’t even solder to program it! I used some hook clips I [...]

Hi acalatrava,
I’m digging up an old thread.
Could you post the link to the equipment you use for soldering again? (old one isn’t active.)
I’ve tried soldering, but unfortunately my skills aren’t up to working with such small pins yet.