acalatrava/openhaystack-firmware #13: Benefits of this alternative firmware?

digitaltrousers, 2022-06-26:
Hi. Sorry if this is obvious or answered elsewhere, but I'm wondering if you could explain the benefits of this fork? I suspect this is the firmware I want, as it seems to be easier to flash via openOCD/SWD. I also see you're using the Nordic Softdevice, which implies that the original doesn't. I'm curious about the benefits of this?
Also if I were to flash this via SWD, would I flash it at address 0x0?
Thanks,
Dan.

acalatrava, 2022-06-26:
Hi! It’s stated on the [Readme file](https://github.com/acalatrava/openhaystack-firmware/blob/main/README.md), basically by using the Softdevice will use a lot less battery than the original one, at least at the moment when I wrote this, I’m not sure if they improved the original firmware to optimize battery power.

digitaltrousers, 2022-06-30:
Thanks. And sorry - I can't believe I missed that!

closed 2022-06-30 by digitaltrousers