acalatrava/openhaystack-firmware #14: Getting Beacon into OpenHaystack.app – right procedure?

henriklied, 2022-07-01:
I'm trying to get some Aliexpress beacons into OpenHaystack.app, and was wondering if I'm doing it right.
Here's my procedure as of now:

1. Deploy a new NRF-accessory in OpenHaystack.app
2. Right click on the new accessory, Copy Advertising Key -> Byte Array
3. Edit main.c, swap out the string for `{byte array}`
4. `NRF_MODEL=nrf51 BOARD=BOARD_ALIEXPRESS make build`
5. Flash the beacon using openocd

There are no errors in this process, but I'm still not getting any location pings in OpenHaystack.app.

Any tips?

mowtschan, 2022-07-01:
I would recommend you just to patch the binary :-) 

[from https://github.com/acalatrava/openhaystack-firmware/blob/main/apps/openhaystack-alternative/README.md](https://github.com/acalatrava/openhaystack-firmware/blob/main/apps/openhaystack-alternative/README.md)

```
NRF_MODEL=nrf51 BOARD=BOARD_ALIEXPRESS ADV_KEY_BASE64=YOUR_ADVERTISEMENT_KEY make patch
```

acalatrava, 2023-01-19:
closed due to inactivity, reopen if needed

closed 2023-01-19 by acalatrava

stmikul, 2023-02-23:
@acalatrava, what is the specific process for patching the binary?