acalatrava/openhaystack-firmware #15: Firmware not working

bruvv, 2022-08-14:
So I have 4 NRF51822 Bluetooth modules and flashed 2 of them with the original openhaystack firmware which worked fine and flashed 2 with the following command:

`NRF_MODEL=nrf51 BOARD=BOARD_ALIEXPRESS ADV_KEY_BASE64=*MYBASE64ADVKEY* make build`

And one with the simple board to test the difference between the crystal build
`NRF_MODEL=nrf51 BOARD=BOARD=BOARD_SIMPLE ADV_KEY_BASE64=*MYBASE64ADVKEY* make build`

But both of them fail to show up in OHS:
[image]

Itheras, 2022-08-19:
what the output file size ? is it a hex or a bin? how are flashing it.

bruvv, 2022-08-19:
Thanks @Itheras ! I am flashing the bin file which is 117.048 bytes (119 KB on disk). The commands I am using to flash it I use: `openocd -f /usr/local/share/openocd/scripts/interface/stlink.cfg -f /usr/local/share/openocd/scripts/target/nrf51.cfg` then telnet to it and execute the followings:
```
halt
nrf51 mass_erase
program /Users/nivong/Documents/scripts/openhaystack-firmware/apps/openhaystack-alternative/compiled/nrf51_firmware.bin
resume
```

I get as an output when flashing:

```
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Adding extra erase range, 0x0001cc70 .. 0x0001cfff
** Programming Finished **
```

Itheras, 2022-08-19:
Try this delete everything in the output . then use this command. NRF_MODEL=nrf51 make build after it finish building sent this command NRF_MODEL=nrf51 BOARD=BOARD_ALIEXPRESS ADV_KEY_BASE64=YOUR_ADVERTISEMENT_KEY make patch  . then flash as before but use the bin that says patched.

Itheras, 2022-08-19:
also after finish flashing issue command  resume and tell me if it halts again by itself.

bruvv, 2022-08-19:
Thanks @Itheras for thinking along. I did the following:
```
sudo rm -r openhaystack-firmware
git clone https://github.com/acalatrava/openhaystack-firmware.git
cd openhaystack-firmware
git submodule update --init
cd into the apps folder and ran commands
```

`NRF_MODEL=nrf51 make build`
<details>
  <summary>Log file</summary>
  
```
BUILD OPTIONS:
  SoftDevice  s130
  SDK         11
  nRF         nrf51822
  RAM         16 kB
  FLASH       256 kB
  Board       BOARD_SIMPLE

rm -rf ../../nrf5x-base/make/../dfu/_build
rm -rf _build/
rm -f *.jlink
rm -f JLink.log
rm -f .gdbinit
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/toolchain/system_nrf51.c
  CC        ble_stack.c
  CC        main.c
  CC        openhaystack.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/softdevice/common/softdevice_handler/softdevice_handler.c
In function 'intern_softdevice_events_execute',
    inlined from 'intern_softdevice_events_execute' at ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/softdevice/common/softdevice_handler/softdevice_handler.c:109:6:
../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/softdevice/common/softdevice_handler/softdevice_handler.c:159:17: warning: 'evt_id' may be used uninitialized [-Wmaybe-uninitialized]
  159 |                 m_sys_evt_handler(evt_id);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/softdevice/common/softdevice_handler/softdevice_handler.c: In function 'intern_softdevice_events_execute':
../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/softdevice/common/softdevice_handler/softdevice_handler.c:136:22: note: 'evt_id' declared here
  136 |             uint32_t evt_id;
      |                      ^~~~~~
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_util_platform.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/common/nrf_drv_common.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/delay/nrf_delay.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_error.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_error_weak.c
  AS        ../../nrf5x-base/make/../startup/startup_nrf51.s
  LD        _build/openhaystack-alternative_s130.elf
  HEX       _build/openhaystack-alternative_s130.hex
  BIN       _build/openhaystack-alternative_s130.bin
  LST       _build/openhaystack-alternative_s130.lst
  SIZE      _build/openhaystack-alternative_s130.elf
   text	   data	    bss	    dec	    hex	filename
   4280	   2176	    160	   6616	   19d8	_build/openhaystack-alternative_s130.elf
mkdir -p compiled
mergehex -m ../../nrf5x-base/sdk/nrf51_sdk_11.0.0/components/softdevice/s130/hex/s130_nrf51_2.0.0_softdevice.hex _build/openhaystack-alternative_s130.hex -o compiled/nrf51_firmware.hex
Parsing input files.
Merging file "s130_nrf51_2.0.0_softdevice.hex" into output.
Merging file "openhaystack-alternative_s130.hex" into output.
Storing merged file.
objcopy --input-target=ihex --output-target=binary compiled/nrf51_firmware.hex compiled/nrf51_firmware.bin
```
</details>

Then `NRF_MODEL=nrf51 BOARD=BOARD_ALIEXPRESS ADV_KEY_BASE64=YOUR_ADVERTISEMENT_KEY make patch`

<details>
  <summary>Log file</summary>
  
```
BUILD OPTIONS:
  SoftDevice  s130
  SDK         11
  nRF         nrf51822
  RAM         16 kB
  FLASH       256 kB
  Board       BOARD_ALIEXPRESS

xxd -p -c 1000000 < compiled/nrf51_firmware.bin | \
		sed 's/4f46464c494e4546494e44494e475055424c49434b45594845524521/60e511fc00d51114c8484304353fca/' | \
		xxd -r -p > compiled/nrf51_firmware_patched.bin
```
</details>

The patched bin file is 117.035 bytes (119 KB on disk). I then flashed with the previous commands

<details>
  <summary>Log file</summary>
 
```
```
</details>

Itheras, 2022-08-19:
still doesn't nothing right ? could you send me your unpatched bin ?

bruvv, 2022-08-19:
Yea it is not booting as far as I can see with BlueSe
[nrf51_firmware.bin.zip](https://github.com/acalatrava/openhaystack-firmware/files/9383190/nrf51_firmware.bin.zip)
e
and this is a photo of the board:
[image]

Itheras, 2022-08-19:
ok yeah I see try this replace the unpatched bin with this one make sure is a bin again lol. then issue  NRF_MODEL=nrf51 BOARD=BOARD_ALIEXPRESS ADV_KEY_BASE64=YOUR_ADVERTISEMENT_KEY make patch    again
[nrf51_firmware.bin.zip](https://github.com/acalatrava/openhaystack-firmware/files/9383236/nrf51_firmware.bin.zip)

Itheras, 2022-08-19:
if it works then your version of arm-none-eabi-objcopy is the possible problem.

mowtschan, 2022-08-19:
@bruvv , I assume that you are aware about some board configuration because you already commented my [PR](https://github.com/acalatrava/openhaystack-firmware/pull/8), just be sure you have correct settings, at least in my case it was a problem, this why I created PR for aliexpress board.

Itheras, 2022-08-19:
I compared his build for aliexpress board  and its different . I had that issue trying to build for the nrf52810 I broke my head for days it ended being my version of mergehex lol . was messing up the softdevice.

bruvv, 2022-08-19:
@Itheras thanks for noticing, I just flashed your nrf51 firmware since I am missing the crystal, thanks @mowtschan for also pitching in. As far as I know, when using `BOARD=BOARD_SIMPLE` That is meant for my aliexpress board but without the crystal.

Itheras, 2022-08-19:
fingers crossed let me know.

mowtschan, 2022-08-19:
and just as a reminder: [from readme](https://github.com/acalatrava/openhaystack-firmware/tree/main/apps/openhaystack-alternative)
```
In case you can't or don't want to build the firmware, you can just patch existing firmware with your advertisement key from OpenHaystack app:

NRF_MODEL=nrf51 BOARD=BOARD_ALIEXPRESS ADV_KEY_BASE64=YOUR_ADVERTISEMENT_KEY make patch
```

Itheras, 2022-08-19:
> and just as a reminder: [from readme](https://github.com/acalatrava/openhaystack-firmware/tree/main/apps/openhaystack-alternative)
>
> ```
> In case you can't or don't want to build the firmware, you can [...]

Yep that's true too. The version of the bin I gave him is not acalatravas version I found this one slightly more energy efficient and is based on [[]](https://github.com/dakhnod/FakeTag)(url) faketag

bruvv, 2022-08-19:
Alright that did not work, I redid everything. Removed and recloned the repo and did nothing. I just flashed the bin file that is in the repo, since I know that that works because the openhaystack bin file I can flash with success. But that did not work either. It is just not visible in the BlueSee app on my macbook. When I flash the exact same hardware with the openhaystack firmware it is visible in BlueSee:
[image]
So what is going on here...

Itheras, 2022-08-19:
mmm just to ask because I have done this but what base64 key are you copying, keyID or advertisement key ?

bruvv, 2022-08-19:
I am not copying anything, I just flashed the bin file without patching it. So I just flashed this bin: https://github.com/acalatrava/openhaystack-firmware/blob/main/apps/openhaystack-alternative/compiled/nrf51_firmware.bin

Itheras, 2022-08-19:
I see let me think. now it became strange. when flashing the unmodified bin to you still get this response at the beginning of the flash.?

Padding image section 0 at 0x0001c92b with 1 bytes

mowtschan, 2022-08-19:
@bruvv, can you try to remove power from the board for some seconds after you flash it, then power it again and then try to search it again with BlueSee?

bruvv, 2022-08-19:
@Itheras 
```
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Adding extra erase range, 0x0001cc70 .. 0x0001cfff
** Programming Finished **
```

@mowtschan did that after running resume, i waited 5 seconds, took the power off for 60 seconds, and powered it and waited 60 seconds. It did show up! So then I just removed all the casks that got installed and reinstalled them all.
```
binutils 2.39
open-ocd 0.11.0
gcc-arm-embedded 11.2-2022.02
nordic-nrf-command-line-tools 10.13.0
segger-jlink 7.70
```

I am now going to try and follow your tips @mowtschan

Itheras, 2022-08-19:
Nice!! good that you got to work thank you @mowtschan

bruvv, 2022-08-19:
Been 2 minutes and it is still not showing up :( so something is wrong with my way of building the file with your config @mowtschan

Itheras, 2022-08-19:
compare your unpatched bin with the prebuilt in the repo. yours did look weird to me.

bruvv, 2022-08-19:
@Itheras how can I compare these? I am pretty new to C++

Itheras, 2022-08-19:
I just use hex fiend It will highlight all the differences. open both files with it and it has an option to compare. there must be an easier way but that what I do. really crude but just check size differences.

bruvv, 2022-08-19:
WAIT WUTH, I just redid everything again, and just issued: `NRF_MODEL=nrf51 BOARD=BOARD_ALIEXPRESS ADV_KEY_BASE64=*KEY* make patch` with the patched board from @mowtschan and my own key this time I made sure the thing was off by making a small short on the board (just touching the + with the -) and now it is showing up!

Itheras, 2022-08-19:
I wouldn't question it hahahaha. Niceee

bruvv, 2022-08-19:
Thanks a lot @Itheras and @mowtschan learned a lot :) Going to close this one and enjoy my tags

closed 2022-08-19 by bruvv

j-paulus, 2022-09-03:
As I'm struggling with the same problem now, what exactly was the solution?

I have a couple of [these](https://www.amazon.de/-/en/dp/B07XNXP28F/) boards. The original OpenHaystack Micro:bit binary works out of the box, but this optimized one doesn't show up in any scanner. I've tried the programming already in all the ways I could imagine, even with a fully re-compiled FW.

The flashing seems to work ok:
```
> halt
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x00011480 msp: 0x20003fd0
> nrf51 mass_erase                                                                                                                    
nRF51822-QFAA(build code: CA/C0) 256kB Flash, 16kB RAM
Mass erase completed.
A reset or power cycle is required if the flash was protected before.
> reset
> halt
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0xc1000003 pc: 0xfffffffe msp: 0xffffffd8
> program /Users/name/Downloads/openhaystack/openhaystack-firmware/apps/openhaystack-alternative/compiled/nrf51_firmware.bin verify
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Adding extra erase range, 0x0001c960 .. 0x0001cbff
** Programming Finished **
** Verify Started **
** Verified OK **
> resume
> reset
```
Omitting the reset after erase doesn't make a difference. Power cycling after erase or after programming does not make a difference.
Exactly the same steps with the OHS FW and the beacon is visible almost immediately.

The interesting thing is that [these](https://www.amazon.de/-/en/dp/B07NN4CB8W) tags work with the patched firmware and the same programming steps nicely.

If it makes any difference, I'm using BOARD_SIMPLE.

mowtschan, 2022-09-04:
@j-paulus try this:

1. change config of the board as it is in my [PR](https://github.com/acalatrava/openhaystack-firmware/pull/8)
2. then build
   (2.1. then patch)
3. power off/on

bruvv, 2022-09-04:
@j-paulus what did the trick for me is to reset the hardware with a little short, so connect the+ to the-. Or you have to wait a few minutes since there is still power left in the device

j-paulus, 2022-09-04:
@mowtschan As far as I understand, your PR is about using the internal RC instead of the external XTAL. The contents of "simple_board.h" that I have are matching your version.  Compiling this results into
```openhaystack-alternative % PATH="/usr/local/opt/binutils/bin:$PATH" NRF_MODEL=nrf51 BOARD=BOARD_SIMPLE ADV_KEY_BASE64=<BASE64_AD_KEY> make build
BUILD OPTIONS:
  SoftDevice  s130
  SDK         11
  nRF         nrf51822
  RAM         16 kB
  FLASH       256 kB
  Board       BOARD_SIMPLE
 
rm -rf ../../nrf5x-base/make/../dfu/_build
rm -rf _build/
rm -f *.jlink
rm -f JLink.log
rm -f .gdbinit
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/toolchain/system_nrf51.c
  CC        ble_stack.c
  CC        main.c
  CC        openhaystack.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/softdevice/common/softdevice_handler/softdevice_handler.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_util_platform.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/common/nrf_drv_common.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/drivers_nrf/delay/nrf_delay.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_error.c
  CC        ../../nrf5x-base/make/../sdk/nrf51_sdk_11.0.0/components/libraries/util/app_error_weak.c
  AS        ../../nrf5x-base/make/../startup/startup_nrf51.s
  LD        _build/openhaystack-alternative_s130.elf
/Applications/ArmGNUToolchain/11.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/11.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc.a(libc_a-closer.o): in function `_close_r':
/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/closer.c:47: warning: _close is not implemented and will always fail
/Applications/ArmGNUToolchain/11.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/11.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc.a(libc_a-lseekr.o): in function `_lseek_r':
/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/lseekr.c:49: warning: _lseek is not implemented and will always fail
/Applications/ArmGNUToolchain/11.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/11.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc.a(libc_a-readr.o): in function `_read_r':
/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/readr.c:49: warning: _read is not implemented and will always fail
/Applications/ArmGNUToolchain/11.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/11.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc.a(libc_a-writer.o): in function `_write_r':
/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/writer.c:49: warning: _write is not implemented and will always fail
  HEX       _build/openhaystack-alternative_s130.hex
  BIN       _build/openhaystack-alternative_s130.bin
  LST       _build/openhaystack-alternative_s130.lst
  SIZE      _build/openhaystack-alternative_s130.elf
   text	   data	    bss	    dec	    hex	filename
   5088	   1408	    892	   7388	   1cdc	_build/openhaystack-alternative_s130.elf
mkdir -p compiled
mergehex -m ../../nrf5x-base/sdk/nrf51_sdk_11.0.0/components/softdevice/s130/hex/s130_nrf51_2.0.0_softdevice.hex _build/openhaystack-alternative_s130.hex -o compiled/nrf51_firmware.hex
Parsing input files.
Merging file "s130_nrf51_2.0.0_softdevice.hex" into output.
Merging file "openhaystack-alternative_s130.hex" into output.
Storing merged file.
objcopy --input-target=ihex --output-target=binary compiled/nrf51_firmware.hex compiled/nrf51_firmware.bin
```
Flashing the resulting .bin is effectively the same as earlier: no beacon found.

Those messages about missing implementations are somehow concerning, to be honest.

@bruvv Removing the cables and shorting the power on the module after programming didn't help in my case. Given that the original OHS FW works straight out of the box (tried it again successfully) is suggesting to me that the problem is not necessarily in the flashing or resets, but something in the binary.