acalatrava/openhaystack-firmware #17: flashed my device but cant deploy

jaekim24, 2022-12-29:
Getting this error. Not sure what to do. 
[image]

jaekim24, 2022-12-29:
nevermind! You can right click and manually deploy it

closed 2022-12-29 by jaekim24

stmikul, 2023-02-23:
@jaekim24, can you explain more about what you mean by this? After dragging and dropping the firmware to my microbit v2, what are the next steps?

stmikul, 2023-02-23:
@jaekim24, I dragged the .hex file to my Micro:bit v2, and it is displaying "OpenHaystack" on the LED panel but I'm not sure how to get this to deploy to the OpenHaystack macOS app to connect to the Find My network. Any help would be appreciated.