acalatrava/openhaystack-firmware #30: Aliexpress Beacon OTA update

DragonHunter274, 2024-06-06:
According to the Datasheet of the nrf51822 aliexpress beacons, which are made by holyiot, they support nrf OTA dfu by default

I haven't been able to build an ota package yet, most documentation seems to focus on the nrf52 family

It would be pretty useful not to have to solder swd to every tag

florian-asche, 2024-09-28:
So you successfully used OTA to flash the holyiot? I have some NRF52810, trying to figure out, how to flash them.

quack3d, 2025-01-27:
Any more info on this?

alexhorner, 2025-02-08:
Cannot confirm. Using nRF Connect, after connecting a DFU button should appear next to the Disconnect button, and also the DFU service should appear in the list of available services alongside the UART service. This does not appear on the HolyIOT devices I recently purchased and connected to. Maybe they removed it?

I successfully flashed two units over SWD instead though.